HKUST-Aerial-Robotics/VINS-Fusion #175: the freq of the odometry is wrong with the camera fps when using mult_thread and stereo?

luobodan, 2022-01-18:
Hello! I was struck in the  frequncey of the output odometry when I turn on the mult_thread with stereo or stereo_imu vins-fusion.When I using the 30fps and 15fps camera , it just output the odometry with 15hz and 7.5hz, using rostopic hz . And after the camera open a few time, the delay of the output also begin to appear , I just think the picture was pushed back to the sequence or buffer area and they stack more and more??
how to fix this problem? I just want to free the output freq and make it in sync with the camera fps?

I read the code , " if(bababa%2)"  in estimitor_node,  is this make it wrong?

thanks you very much!

Pallav1299, 2022-02-15:
@luobodan, have a look at [this](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/92#issuecomment-562021646).

luobodan, 2022-02-28:
> @luobodan, have a look at [this](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/92#issuecomment-562021646).

thank you！

closed 2022-02-28 by luobodan