HKUST-Aerial-Robotics/VINS-Fusion #177: installation error when catkin build

PeixuanGao, 2022-01-27:
i tried to install the gpu version and when i build it it shows this

Profile:                     default
Extending:          [cached] /opt/ros/noetic
Workspace:                   /home/peixuan/workspace2
-------------------------------------------------------------
Build Space:        [exists] /home/peixuan/workspace2/build
Devel Space:        [exists] /home/peixuan/workspace2/devel
Install Space:      [unused] /home/peixuan/workspace2/install
Log Space:          [exists] /home/peixuan/workspace2/logs
Source Space:       [exists] /home/peixuan/workspace2/src
DESTDIR:            [unused] None
-------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        None
-------------------------------------------------------------
Additional CMake Args:       None
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
-------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
-------------------------------------------------------------
Workspace configuration appears valid.
-------------------------------------------------------------
[build] Found '4' packages in 0.0 seconds.                                     
[build] Updating package table.                                                
Starting  >>> camera_models                                                    
Starting  >>> global_fusion                                                    
Starting  >>> vins                                                             
_______________________________________________________________________________
Errors     << vins:cmake /home/peixuan/workspace2/logs/vins/build.cmake.000.log
CMake Error at /opt/ros/noetic/share/catkin/cmake/catkinConfig.cmake:83 (find_package):
  Could not find a package configuration file provided by "camera_models"
  with any of the following names:

    camera_modelsConfig.cmake
    camera_models-config.cmake

  Add the installation prefix of "camera_models" to CMAKE_PREFIX_PATH or set
  "camera_models_DIR" to a directory containing one of the above files.  If
  "camera_models" provides a separate development package or SDK, be sure it
  has been installed.
Call Stack (most recent call first):
  CMakeLists.txt:9 (find_package)

cd /home/peixuan/workspace2/build/vins; catkin build --get-env vins | catkin env -si  /usr/bin/cmake /home/peixuan/workspace2/src/VINS-Fusion-gpu/vins_estimator --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/home/peixuan/workspace2/devel/.private/vins -DCMAKE_INSTALL_PREFIX=/home/peixuan/workspace2/install; cd -

...............................................................................
Failed     << vins:cmake                   [ Exited with code 1 ]              
Failed    <<< vins                         [ 4.9 seconds ]                     
Abandoned <<< loop_fusion                  [ Unrelated job failed ]            
_______________________________________________________________________________
Errors     << camera_models:make /home/peixuan/workspace2/logs/camera_models/build.make.001.log
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc: In constructor ‘camodocal::Chessboard::Chessboard(cv::Size, cv::Mat&)’:
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc:20:38: error: ‘CV_GRAY2BGR’ was not declared in this scope
   20 |         cv::cvtColor(image, mSketch, CV_GRAY2BGR);
      |                                      ^~~~~~~~~~~
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc:26:37: error: ‘CV_BGR2GRAY’ was not declared in this scope
   26 |         cv::cvtColor(image, mImage, CV_BGR2GRAY);
      |                                     ^~~~~~~~~~~
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc: In member function ‘void camodocal::Chessboard::findCorners(bool)’:
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc:34:43: error: ‘CV_CALIB_CB_ADAPTIVE_THRESH’ was not declared in this scope
   34 |                                           CV_CALIB_CB_ADAPTIVE_THRESH +
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/chessboard/Chessboard.cc:35:43: error: ‘CV_CALIB_CB_NORMALIZE_IMAGE’ was not declared in this scope
   35 |                                           CV_CALIB_CB_NORMALIZE_IMAGE +
      |                                           ^~~~~~~~~~~~~~~~~~~~~~~~~~~
[... 167 lines cut ...]
  253 |                        5, green, 2, CV_AA, drawShiftBits);
      |                                     ^~~~~
      |                                     CV_MSA
/home/peixuan/workspace2/src/VINS-Fusion-gpu/camera_models/src/calib/CameraCalibration.cc:275:24: error: ‘CV_AA’ was not declared in this scope; did you mean ‘CV_MSA’?
  275 |                     1, CV_AA);
      |                        ^~~~~
      |                        CV_MSA
make[2]: *** [CMakeFiles/camera_models.dir/build.make:76: CMakeFiles/camera_models.dir/src/calib/CameraCalibration.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:157: CMakeFiles/camera_models.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:265: CMakeFiles/Calibrations.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
cd /home/peixuan/workspace2/build/camera_models; catkin build --get-env camera_models | catkin env -si  /usr/bin/make --jobserver-auth=3,4; cd -

...............................................................................
Failed     << camera_models:make           [ Exited with code 2 ]              
Failed    <<< camera_models                [ 1 minute and 8.7 seconds ]        
Finished  <<< global_fusion                [ 1 minute and 20.0 seconds ]       
[build] Summary: 1 of 4 packages succeeded.                                    
[build]   Ignored:   None.                                                     
[build]   Warnings:  None.                                                     
[build]   Abandoned: 1 packages were abandoned.                                
[build]   Failed:    2 packages failed.                                        
[build] Runtime: 1 minute and 20.3 seconds total.                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.

GopiRajuMatta, 2022-10-18:
I am getting similar error. Any update on this?

czp0204, 2023-04-17:
modify the opencv version