HKUST-Aerial-Robotics/VINS-Fusion #182: ZED on VINS-Fusion

mikodham, 2022-03-15:
Following up the issue on https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/issues/151 , my code is very similar to the previous post that has been closed. I also found it strange that the vio trajectory is random, and furthermore there is a huge delay between camera_image (from zed "/zed2/zed_node/left_raw/image_raw_color") and track_image when displayed on rviz. 

And for many times. these error (sometimes/most of the time, sometimes doesnt) comes up. 
[ WARN] [1647340249.513694929]: numerical unstable in preintegration

I have used the calibrator here as well to find the camera parameters for the 2 cameras so I can get projection and distortion parameters. 
https://support.stereolabs.com/hc/en-us/articles/360012749113-How-can-I-use-Kalibr-with-the-ZED-Mini-camera-in-ROS-
But the trajectory produced by VIO is still random. Does anyone know why?

mikodham, 2022-03-17:
To be more precise, I figured that the cause is uncalibrated body_T_cam0 and body_T_cam1
I am still new to this, but does anyone know how to get the 4*4 by matrix in zed. 

On zed2, the topic left_cam_imu_transform is provided, but how to convert this into 4*4 body_T_cam0. 
There is no right_cam_imu_transform, so I am not sure how to get body_T_cam1

rostopic echo /zed2/zed_node/left_cam_imu_transform 
translation: 
  x: -0.00200000009499
  y: -0.02300000377
  z: 0.000220000030822
rotation: 
  x: -0.0001220703125
  y: -0.000244140625
  z: -0.000897029938642
  w: 0.999999582767

mikodham, 2022-03-18:
Must use kalibr. The default parameters provided by Zed ros cant be trusted. 
https://github.com/Robotics-and-Perception-Team/VINS-Fusion-Config-with-Kalibr

closed 2022-03-18 by mikodham

dheerubhai-101, 2023-07-19:
Hi @mikodham , I am trying something similar with a ZED camera. I have used the calibration parameters from the zed calibration file. When running the node, the video output in Rviz seems fine and it can detect the features in that video but there is no World to Body transform occurring. Consequently, there is nothing visualized in RViz. I am using a bag file with only the left and right video streams recorded.  Can you please let me know how to tackle this?