HKUST-Aerial-Robotics/VINS-Fusion #185: calcOpticalFlowPyrLK causing assertion fail(BUG)

stevedanomodolor, 2022-04-01:
My system is ubuntu 20.4 with opencv 4. 
I am trying to run the euroc example but everytime I get the following error
```
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.2.0) /home/stevedan/libraries/opencv/modules/video/src/lkpyramid.cpp:1257: error: (-215:Assertion failed) maxLevel >= 0 && winSize.width > 2 && winSize.height > 2 in function 'calc'
```
I was able to trace the problem to the feature tracker file but I don't know how to fix it, any suggestion?
in the trackimage function 

```
        else
        {
          std::cout << "**************Here3*****" << std::endl;
          std::cout << prev_img.size() << " " << cur_img.size() << " " << cur_pts.size() <<" " << prev_pts.size() << std::endl;
          cv::calcOpticalFlowPyrLK(prev_img, cur_img, prev_pts, cur_pts, status, err, cv::Size(21, 21), 3);
        }
```

stevedanomodolor, 2022-04-01:
Solves issue, it was something to do with opencv library

closed 2022-04-01 by stevedanomodolor

MadlyFX, 2022-05-14:
How did you wind up solving this?

stevedanomodolor, 2022-05-14:
I believe the reason was that I was sending an empty image, make sure that before you send anything that the size is not zero.

MadlyFX, 2022-05-14:
Interesting, thanks. I'm trying to use a Realsense D435i and running into this error, but I'm definitely sending valid images.

stevedanomodolor, 2022-05-15:
The previus_img size was like empty i think,  print the image size as I did 
else
        {
          std::cout << "**************Here3*****" << std::endl;
          std::cout << prev_img.size() << " " << cur_img.size() << " " << cur_pts.size() <<" " << prev_pts.size() << std::endl;
          cv::calcOpticalFlowPyrLK(prev_img, cur_img, prev_pts, cur_pts, status, err, cv::Size(21, 21), 3);
        }
I think this was the issue, I don't recall exactly