HKUST-Aerial-Robotics/VINS-Fusion #186: Get Covariance of Twist and Pose in vins_estimator/odometry

mikodham, 2022-04-15:
I followed https://github.com/HKUST-Aerial-Robotics/VINS-Mono/issues/74
to get pose covariance, which can then be published in visualization.cpp as topic named `vins_estimator/odometry`
as shown in the figure below. 

On visualization.cpp
[image]
On estimator.cpp, function Estimator::optimization()
[image]

However, any idea how to get twist covariance?

DanielCyberbee, 2024-11-25:
@zxytql i have the same issue. have you managed to fixed it?

zxytql, 2024-11-26:
> @zxytql i have the same issue. have you managed to fixed it?

No, I haven't.