HKUST-Aerial-Robotics/VINS-Fusion #188: error when I run 'loop_fusion'

XFFer, 2022-05-14:
Why the 'loop_fusion_node' will be killed when a loop is detected?
Some errors in terminator:

config_file: vins_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml
vocabulary_file/home/fengyx/vins_ws/src/VINS-Fusion/loop_fusion/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/fengyx/vins_ws/src/VINS-Fusion/loop_fusion/../support_files/brief_pattern.yml
cam calib path: vins_ws/src/VINS-Fusion/config/euroc/cam0_mei.yaml
VIO input, perfrom 4 DoF (x, y, z, yaw) pose graph optimization
no previous pose graph
optimize pose graph 
terminate called after throwing an instance of 'std::system_error'
  what():  Invalid argument
Aborted (core dumped)

HuangKangJin, 2022-09-14:
I have the same error, have you solved this problem?

XFFer, 2022-09-15:
> I have the same error, have you solved this problem?

no, I didn't dig out the solution.