HKUST-Aerial-Robotics/VINS-Fusion #193: About demo of stereo camera and IMU fusion of KITTI data sets

987428377, 2022-08-07:
Dear master： 

    I tried to run KitI's stereo camera 、 IMU 、GPS with vins algorithm, but the effect was not obvious. I calculated the conversion matrix from IMU to camera according to Kitti's calibration as follows:

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [0, -1, 0, -0.8086759,
          0, 0, -1, 0.3195559,
          1, 0, 0, -1.0667645,
          0, 0, 0, 1]

body_T_cam1: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [0, -1, 0, -0.8086759,
          0, 0, -1, 0.319559,
          1, 0, 0, -0.5297645,
          0, 0, 0, 1]

The code I added to the IMU data is as follows:
                        Vector3d linear_acceleration(ax,ay,az);
			Vector3d angular_velocity(wx,wy,wz);
			estimator.inputIMU(imgTime, linear_acceleration, angular_velocity);

I tried to rotate the rotation matrix 90 degrees around the X 、 Y 、 Z axis, but it still didn't work. I am very depressed. Can you help me? Thank you very much and look forward to your reply！

Who can help me?

987428377, 2022-08-07:
The dataset is        2011_ 09_ 29_ drive_ 0004_ sync

vjsliogeris, 2023-11-23:
I am also very depressed.