HKUST-Aerial-Robotics/VINS-Fusion #194: catkin_make完之后拿Euroc数据集测试，但是报错。

GGshmily, 2022-08-23:
数据集是:MH_02_easy.bag
terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.5-dev) /home/jimlee/LHG/opencv/modules/core/src/matrix_wrap.cpp:423: error: (-215:Assertion failed) i < 0 in function 'size'

已放弃 (核心已转储)

chongjeelee, 2022-09-20:
i have the same problem ,do you solved it ?

GopiRajuMatta, 2022-10-18:
Facing the same problem. Any update on this?

GGshmily, 2022-10-18:
> 你是不是之前的编译出现了警告？[ WARN] [1665062148.295888966]: fix extrinsic param 。我用了ulimit -s 102400扩充栈内存后就变成新问题了。

我解决了，是我的电脑里有两个版本的opencv，我拿opencv4.5编译的，但是运行的时候会使用libopencv3的库，所以会崩溃

GGshmily, 2022-10-18:
maybe is the contradictory of opencv4.5 and libopencv3. I build with opencv4.5，but VINS automaticly used libopencv3. You can  change libopencv3 to libopencv4.5 or build VINS with opencv3.

GopiRajuMatta, 2022-10-18:
Thank you for the response. Once I built it using opencv4, how can I change it to opencv3? Can you please help me with this.