HKUST-Aerial-Robotics/VINS-Fusion #207: VINS-Fusion on car demonstration ubuntu 18.04

czp0204, 2023-04-23:
Hi, newbie here. I just download the VINS-Fusion and car.bag and I want to run the car demonstration. Then I encountered this problem:
VINS_RESULT_PATH not opened! Check if the path exists.

Here is the log.
```
czp@czp02:~/Downloads/vins$ rosrun vins vins_node ~/Downloads/vins/src/VINS-Fusion/config/vi_car/vi_car.yaml
[ INFO] [1682259910.761505858]: init begins
config_file: /home/czp/Downloads/vins/src/VINS-Fusion/config/vi_car/vi_car.yaml
USE_IMU: 1
IMU_TOPIC: /imu0
result path /home/tong/output/stamped_traj_estimate.txt
VINS_RESULT_PATH not opened! Check if the path exists.
[ WARN] [1682259910.768663156]:  fix extrinsic param 
camera number 2
[ INFO] [1682259910.769451587]: Synchronized sensors, fix time offset: 0
[ INFO] [1682259910.769469371]: ROW: 480 COL: 752 
 exitrinsic cam 0
    0.99991   0.0119488 -0.00604042
 -0.0119315    0.999925  0.00288942
 0.00607449 -0.00281709    0.999978
  0.0388369 -0.00474068 -0.00359977
 exitrinsic cam 1
    0.999935    0.0107942   0.00360279
  -0.0107968     0.999941  0.000698778
 -0.00359504 -0.000737631     0.999993
 -0.0718311 0.000219524 -0.00719528
set g   0   0 9.8
[ INFO] [1682259910.769566572]: reading paramerter of camera /home/czp/Downloads/vins/src/VINS-Fusion/config/vi_car/cam0_mei.yaml
[ INFO] [1682259910.769671035]: reading paramerter of camera /home/czp/Downloads/vins/src/VINS-Fusion/config/vi_car/cam1_mei.yaml
MULTIPLE_THREAD is 1
[ WARN] [1682259910.769757077]: waiting for image and imu...
init first imu pose
averge acc -0.238436 -9.751367 0.113449
init R0 
   0.999701   -0.024441 0.000284351
          0   0.0116334    0.999932
 -0.0244426   -0.999634   0.0116299
[ WARN] [1682259915.747214036]: gyroscope bias initial calibration   0.0133774 -0.00643623   0.0112976
[ INFO] [1682259915.791335916]: Initialization finish!
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists.
VINS_RESULT_PATH not opened! Check if the path exists
and then persistently printed this
```
ROS node was launched with following comands
`roslaunch vins vins_rviz.launch
rosrun vins vins_node ~/Downloads/vins/src/VINS-Fusion/config/vi_car/vi_car.yaml
rosbag play car.bag`

czp0204, 2023-04-23:
I didn't change any yaml files in config/vi_car

peitfff, 2025-04-22:
Fill in your own path for "output_path:" in a configuration file (.yaml) in the code.