HKUST-Aerial-Robotics/VINS-Fusion #209: GoPro08

SnkFst, 2023-06-01:
Hello, 

i used https://github.com/AutonomousFieldRoboticsLab/gopro_ros to create out of an MP4 file a bag for VINS (it has images and imu data). After i fixed some issues with the topics and types to the right one (compressed_images to normal images in my bag file) and wrote a new config file (so topics for vins and my bag match) i played my bag with vins . It works, so my imu and images are read but in my rviz windows it only shows the pictures and not the map with pointcloud. i don't know whtas the issue because i checked everything an it seems right. 
My bag fil:  rosbag info converted.bag
path:        converted.bag
version:     2.0
duration:    27.0s
start:       May 04 2023 20:47:40.11 (1683226060.11)
end:         May 04 2023 20:48:07.13 (1683226087.13)
size:        4.7 GB
messages:    6995
compression: none [1621/1621 chunks]
types:       sensor_msgs/Image [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu   [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /gopro/image_raw/compressed   1620 msgs    : sensor_msgs/Image
             /gopro/imu                    5375 msgs    : sensor_msgs/Imu
------
my config file: %YAML:1.0

#common parameters
#support: 1 imu 1 cam; 1 imu 2 cam: 2 cam; 
imu: 1         
num_of_cam: 1  

imu_topic: "/gopro/imu"
image0_topic: "/gopro/image_raw/compressed"
output_path: "~/output/"

cam0_calib: "cam0_GoPro8.yaml"
image_width: 2704
image_height: 1520
   

# Extrinsic parameter between IMU and Camera.
estimate_extrinsic: 0   # 0  Have an accurate extrinsic parameters. We will trust the following imu^R_cam, imu^T_cam, don't change it.
                        # 1  Have an initial guess about extrinsic parameters. We will optimize around your initial guess.

body_T_cam0: !!opencv-matrix
   rows: 4
   cols: 4
   dt: d
   data: [1, 0, 0, 0,
           0, 1, 0,  0,
          0, 0, 1, 0,
           0, 0, 0, 1]

#Multiple thread support
multiple_thread: 1

#feature traker paprameters
max_cnt: 150            # max feature number in feature tracking
min_dist: 30            # min distance between two features 
freq: 10                # frequence (Hz) of publish tracking result. At least 10Hz for good estimation. If set 0, the frequence will be same as raw image 
F_threshold: 1.0        # ransac threshold (pixel)
show_track: 1           # publish tracking image as topic
flow_back: 1            # perform forward and backward optical flow to improve feature tracking accuracy

#optimization parameters
max_solver_time: 0.04  # max solver itration time (ms), to guarantee real time
max_num_iterations: 8   # max solver itrations, to guarantee real time
keyframe_parallax: 10.0 # keyframe selection threshold (pixel)

#imu parameters       The more accurate parameters you provide, the better performance
acc_n: 0.1          # accelerometer measurement noise standard deviation. 
gyr_n: 0.01         # gyroscope measurement noise standard deviation.     
acc_w: 0.001        # accelerometer bias random work noise standard deviation.  
gyr_w: 0.0001       # gyroscope bias random work noise standard deviation.     
g_norm: 9.81007     # gravity magnitude

#unsynchronization parameters
estimate_td: 0                      # online estimate time offset between camera and imu
td: 0.0                             # initial value of time offset. unit: s. readed image clock + td = real image clock (IMU clock)

#loop closure parameters
load_previous_pose_graph: 0        # load and reuse previous pose graph; load from 'pose_graph_save_path'
pose_graph_save_path: "~/output/pose_graph/" # save and load path
save_image: 1                   # save image in pose graph for visualization prupose; you can close this function by setting 0 
-------------------------------------------------

So that are the important information. If you can help i would be very grateful. 
Greetings

SnkFst, 2023-06-02:
I still have the same issue, but i added the leica topic, so my bag look like this: 
```bash
path:        final2.bag
version:     2.0
duration:    27.0s
start:       May 04 2023 20:47:40.11 (1683226060.11)
end:         May 04 2023 20:48:07.13 (1683226087.13)
size:        4.7 GB
messages:    12529
compression: none [1621/1621 chunks]
types:       geometry_msgs/PointStamped [c63aecb41bfdfd6b7e1fac37c7cbe7bf]
             sensor_msgs/Image          [060021388200f6f0f447d0fcd9c64743]
             sensor_msgs/Imu            [6a62c6daae103f4ff57a132d6f95cec2]
topics:      /cam0/image_raw   1620 msgs    : sensor_msgs/Image
             /imu0             5375 msgs    : sensor_msgs/Imu
             /leica/position   5534 msgs    : geometry_msgs/PointStamped
´´´
Every important data should be in this bag and it still creates no pointcloud.

sajjad-hm, 2023-08-01:
@SnkFst Could you solve this issue?

nazmulmunna07, 2023-08-02:
@SnkFst Exciting to know we're on similar projects! Have you managed to generate a point cloud?  If you have, I would truly appreciate it if you could generously share your config YAML file.

SnkFst, 2023-08-03:
@sajjad-hm @nazmulmunna07 unfortunately i didn't solve the problem.

sajjad-hm, 2023-08-06:
> @sajjad-hm @nazmulmunna07 unfortunately i didn't solve the problem.

@SnkFst 
I'm trying using Android phone. I have calibrated IMU+CAMERA combined and got all the calibration data. 

Now I'm having trouble to figure out changing the parameters using my calibration data on the any config.yaml file. 

Some of the parameters between my calibration result and existing config files in the vins mono config folder seems confusing to me.