HKUST-Aerial-Robotics/VINS-Fusion #212: 我启动vins后内存一直在增加直到占满，是不是哪里没写好内存溢出了啊？

yst-one, 2023-06-15:


ZW-PRO, 2023-12-26:
Have you solved this problem?

JanekDev, 2024-02-13:
@ZW-PRO any updates on the problem? I have some minor observations, so we could possibly exchange info and work something out?

ZW-PRO, 2024-02-17:
> @ZW-PRO any updates on the problem? I have some minor observations, so we could possibly exchange info and work something out?

Yes. I'm running VINS in ros2 and I've noticed that when I limit the size of the queue in the img0_callback() function, the memory doesn't increase anymore, but I guess this can cause issues like image loss. The probable reason is that the image processing speed in the queue cannot keep up with the image speed of the incoming queue.

ZW-PRO, 2024-08-29:
> Referenc

Have you solved this problem?

ZW-PRO, 2024-08-29:
@JanekDev Have you solved this problem?