HKUST-Aerial-Robotics/VINS-Fusion #215: Docker build errors on ubuntu 20.04

TakShimoda, 2023-07-11:
I'm on Ubuntu 20.04 with ros noetic and docker desktop for linux and tried installing on docker as per section 7 on README.md. 

After typing the `make build` command, I get the following error when it's trying to install ceres. 

```
[+] Building 248.9s (6/11)                                                                                                    docker:desktop-linux
 => [internal] load .dockerignore                                                                                                             0.0s
 => => transferring context: 2B                                                                                                               0.0s
 => [internal] load build definition from Dockerfile                                                                                          0.0s
 => => transferring dockerfile: 1.90kB                                                                                                        0.0s
 => WARN: Empty continuation line found in: RUN   if [ "x$(nproc)" = "x1" ] ; then export USE_PROC=1 ;       else export USE_PROC=$(($(nproc  0.0s
 => [internal] load metadata for docker.io/library/ros:melodic-perception                                                                     1.2s
 => [1/7] FROM docker.io/library/ros:melodic-perception@sha256:d86bda10e02495c7d810d9f95d8f988c3cccfb937c704cf2ebb08c0255de975b              61.3s
 => => resolve docker.io/library/ros:melodic-perception@sha256:d86bda10e02495c7d810d9f95d8f988c3cccfb937c704cf2ebb08c0255de975b               0.0s
 => => sha256:d86bda10e02495c7d810d9f95d8f988c3cccfb937c704cf2ebb08c0255de975b 776B / 776B                                                    0.0s
 => => sha256:02cd7ca284b46b592d1df136ac43eec656787e71da74a06378fc68b9955aa91a 2.63kB / 2.63kB                                                0.0s
 => => sha256:41af1b5f0f51947706ae712999cf098bef968a7799e7cb4bb2268829e62a6ab3 26.72MB / 26.72MB                                              7.7s
 => => sha256:ce8c91ad7650f55a57d952458067fa5d43fc42f60aa1300eac0321468efd7f1d 4.88MB / 4.88MB                                                3.1s
 => => sha256:5065596e2d21b46202eed81d2b8fa9eade80dcaeef7445f085b6fb2b98569611 5.87kB / 5.87kB                                                0.0s
 => => sha256:bc27456075680427d55e79ddce0ccec278d1f81ab5181d13386de5ac3f085b90 818.91kB / 818.91kB                                            0.5s
 => => sha256:0683d606786339b4368e89c6aaf1c72169b6aa7e6630aefddaeac503bc6552bc 233B / 233B                                                    0.6s
 => => sha256:e66495e509f4c45b21e332a761ce4b1d4f207417e3e0b51b3c71dad8e9697cce 2.41kB / 2.41kB                                                0.7s
 => => sha256:4c731a0490674b377c6cd43c7d5012868bf98a22b30417a9bbbc9f016d73d07c 259.61MB / 259.61MB                                           40.6s
 => => sha256:25f9d35f26e100c81e0e5c8ceecdf6a25d7beb14187fb876d869b81530cf831a 194B / 194B                                                    3.2s
 => => sha256:c417956853271dd45a847801f29dc2a06fe7d594a677a67abef90a7d63b95806 70.46MB / 70.46MB                                             26.0s
 => => sha256:d9a12ca12306ba5f87d1bba2dcd5e4f882dff48c2cae9e526cd23657a941d829 284.82kB / 284.82kB                                            7.9s
 => => extracting sha256:41af1b5f0f51947706ae712999cf098bef968a7799e7cb4bb2268829e62a6ab3                                                     0.6s
 => => sha256:62fc4b75f8dc61a781d06a268987b8cbac8c0df0fc2e30457cfc6f0a23d9e032 75.00MB / 75.00MB                                             18.5s
 => => extracting sha256:bc27456075680427d55e79ddce0ccec278d1f81ab5181d13386de5ac3f085b90                                                     0.2s
 => => extracting sha256:ce8c91ad7650f55a57d952458067fa5d43fc42f60aa1300eac0321468efd7f1d                                                     0.1s
 => => extracting sha256:0683d606786339b4368e89c6aaf1c72169b6aa7e6630aefddaeac503bc6552bc                                                     0.0s
 => => extracting sha256:e66495e509f4c45b21e332a761ce4b1d4f207417e3e0b51b3c71dad8e9697cce                                                     0.0s
 => => sha256:6a32c2945c91a64bbd820b82a01cef84b2d6bfa42ba78324c9f212667d1bbea7 305.70MB / 305.70MB                                           51.7s
 => => extracting sha256:4c731a0490674b377c6cd43c7d5012868bf98a22b30417a9bbbc9f016d73d07c                                                     7.9s
 => => extracting sha256:25f9d35f26e100c81e0e5c8ceecdf6a25d7beb14187fb876d869b81530cf831a                                                     0.0s
 => => extracting sha256:c417956853271dd45a847801f29dc2a06fe7d594a677a67abef90a7d63b95806                                                     1.2s
 => => extracting sha256:d9a12ca12306ba5f87d1bba2dcd5e4f882dff48c2cae9e526cd23657a941d829                                                     0.0s
 => => extracting sha256:62fc4b75f8dc61a781d06a268987b8cbac8c0df0fc2e30457cfc6f0a23d9e032                                                     2.6s
 => => extracting sha256:6a32c2945c91a64bbd820b82a01cef84b2d6bfa42ba78324c9f212667d1bbea7                                                     7.7s
 => [internal] load build context                                                                                                             1.2s
 => => transferring context: 210.26MB                                                                                                         1.2s
 => ERROR [2/7] RUN   if [ "x$(nproc)" = "x1" ] ; then export USE_PROC=1 ;       else export USE_PROC=$(($(nproc)/2)) ; fi &&       apt-ge  186.4s
------
 > [2/7] RUN   if [ "x$(nproc)" = "x1" ] ; then export USE_PROC=1 ;       else export USE_PROC=$(($(nproc)/2)) ; fi &&       apt-get update && apt-get install -y       cmake       libatlas-base-dev       libeigen3-dev       libgoogle-glog-dev       libsuitesparse-dev       python-catkin-tools       ros-melodic-cv-bridge       ros-melodic-image-transport       ros-melodic-message-filters       ros-melodic-tf &&       rm -rf /var/lib/apt/lists/* &&       git clone https://ceres-solver.googlesource.com/ceres-solver &&       cd ceres-solver &&       git checkout tags/1.12.0 &&       mkdir build && cd build &&       cmake .. &&       make -j$(USE_PROC) install &&       rm -rf ../../ceres-solver &&       mkdir -p /root/catkin_ws/src/VINS-Fusion/:
0.321 Get:1 http://archive.ubuntu.com/ubuntu bionic InRelease [242 kB]
0.322 Get:2 http://security.ubuntu.com/ubuntu bionic-security InRelease [88.7 kB]
0.363 Get:3 http://snapshots.ros.org/melodic/final/ubuntu bionic InRelease [13.8 kB]
0.536 Get:4 http://snapshots.ros.org/melodic/final/ubuntu bionic/main amd64 Packages [1038 kB]
0.718 Get:5 http://security.ubuntu.com/ubuntu bionic-security/universe amd64 Packages [1636 kB]
0.757 Get:6 http://archive.ubuntu.com/ubuntu bionic-updates InRelease [88.7 kB]
0.857 Get:7 http://archive.ubuntu.com/ubuntu bionic-backports InRelease [83.3 kB]
0.971 Get:8 http://archive.ubuntu.com/ubuntu bionic/main amd64 Packages [1344 kB]
1.249 Get:9 http://security.ubuntu.com/ubuntu bionic-security/multiverse amd64 Packages [23.8 kB]
1.249 Get:10 http://security.ubuntu.com/ubuntu bionic-security/restricted amd64 Packages [1688 kB]
1.336 Get:11 http://archive.ubuntu.com/ubuntu bionic/restricted amd64 Packages [13.5 kB]
1.336 Get:12 http://archive.ubuntu.com/ubuntu bionic/universe amd64 Packages [11.3 MB]
1.576 Get:13 http://security.ubuntu.com/ubuntu bionic-security/main amd64 Packages [3373 kB]
2.552 Get:14 http://archive.ubuntu.com/ubuntu bionic/multiverse amd64 Packages [186 kB]
2.563 Get:15 http://archive.ubuntu.com/ubuntu bionic-updates/universe amd64 Packages [2410 kB]
2.745 Get:16 http://archive.ubuntu.com/ubuntu bionic-updates/restricted amd64 Packages [1728 kB]
2.860 Get:17 http://archive.ubuntu.com/ubuntu bionic-updates/main amd64 Packages [3785 kB]
3.120 Get:18 http://archive.ubuntu.com/ubuntu bionic-updates/multiverse amd64 Packages [30.8 kB]
3.284 Get:19 http://archive.ubuntu.com/ubuntu bionic-backports/main amd64 Packages [64.0 kB]
3.556 Get:20 http://archive.ubuntu.com/ubuntu bionic-backports/universe amd64 Packages [20.6 kB]
3.582 Fetched 29.2 MB in 3s (8481 kB/s)
3.582 Reading package lists...
4.426 Reading package lists...
5.247 Building dependency tree...
5.362 Reading state information...
5.449 libeigen3-dev is already the newest version (3.3.4-4).
[... 680 lines cut ...]
[... cut ...] f &&       rm -rf /var/lib/apt/lists/* &&       git clone https://ceres-solver.googlesource.com/ceres-solver &&       cd ceres-solver &&       git checkout tags/${CERES_VERSION} &&       mkdir build && cd build &&       cmake .. &&       make -j$(USE_PROC) install &&       rm -rf ../../ceres-solver &&       mkdir -p $CATKIN_WS/src/VINS-Fusion/" did not complete successfully: exit code: 2
make: *** [Makefile:13: build] Error 1
```
Any pointers to help with this issue would be greatly appreciated.

closed 2023-10-03 by TakShimoda

Myracle1, 2024-03-07:
According to the previous Dockerfile, I made some little modifications, the modified Dockerfile was below:

FROM osrf/ros:kinetic-desktop-full

ENV CERES_VERSION="1.12.0"
ENV CATKIN_WS=/root/catkin_ws

# set up thread number for building
RUN   apt-get update && apt-get install -y \
      cmake \
      libatlas-base-dev \
      libeigen3-dev \
      libgoogle-glog-dev \
      libsuitesparse-dev \
      python-catkin-tools \
      ros-${ROS_DISTRO}-cv-bridge \
      ros-${ROS_DISTRO}-image-transport \
      ros-${ROS_DISTRO}-message-filters \
      ros-${ROS_DISTRO}-tf && \
      rm -rf /var/lib/apt/lists/* && \
      # Build and install Ceres
      git clone https://gitee.com/mirrors/ceres-solver && \
      cd ceres-solver && \
      git checkout tags/${CERES_VERSION} && \
      mkdir build && cd build && \
      cmake .. && \
      make -j4 install && \
      rm -rf ../../ceres-solver && \
      mkdir -p $CATKIN_WS/src/VINS-Fusion/

# Copy VINS-Fusion
COPY ./ $CATKIN_WS/src/VINS-Fusion/
# use the following line if you only have this dockerfile
# RUN git clone https://github.com/HKUST-Aerial-Robotics/VINS-Fusion.git

# Build VINS-Fusion
WORKDIR $CATKIN_WS
ENV TERM xterm
ENV PYTHONIOENCODING UTF-8
RUN catkin config \
      --extend /opt/ros/$ROS_DISTRO \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=Release && \
    catkin build && \
    sed -i '/exec "$@"/i \
            source "/root/catkin_ws/devel/setup.bash"' /ros_entrypoint.sh

Myracle1, 2024-03-07:
According to the previous Dockerfile, I made some little modifications, the modified Dockerfile was below:

FROM osrf/ros:kinetic-desktop-full

ENV CERES_VERSION="1.12.0"
ENV CATKIN_WS=/root/catkin_ws

RUN   apt-get update && apt-get install -y \
      cmake \
      libatlas-base-dev \
      libeigen3-dev \
      libgoogle-glog-dev \
      libsuitesparse-dev \
      python-catkin-tools \
      ros-${ROS_DISTRO}-cv-bridge \
      ros-${ROS_DISTRO}-image-transport \
      ros-${ROS_DISTRO}-message-filters \
      ros-${ROS_DISTRO}-tf && \
      rm -rf /var/lib/apt/lists/* && \

      git clone https://gitee.com/mirrors/ceres-solver && \
      cd ceres-solver && \
      git checkout tags/${CERES_VERSION} && \
      mkdir build && cd build && \
      cmake .. && \
      make -j4 install && \
      rm -rf ../../ceres-solver && \
      mkdir -p $CATKIN_WS/src/VINS-Fusion/

COPY ./ $CATKIN_WS/src/VINS-Fusion/

WORKDIR $CATKIN_WS
ENV TERM xterm
ENV PYTHONIOENCODING UTF-8
RUN catkin config \
      --extend /opt/ros/$ROS_DISTRO \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=Release && \
    catkin build && \
    sed -i '/exec "$@"/i \
            source "/root/catkin_ws/devel/setup.bash"' /ros_entrypoint.sh

Myracle1, 2024-03-07:
`FROM osrf/ros:kinetic-desktop-full

ENV CERES_VERSION="1.12.0"
ENV CATKIN_WS=/root/catkin_ws

      # set up thread number for building
RUN   apt-get update && apt-get install -y \
      cmake \
      libatlas-base-dev \
      libeigen3-dev \
      libgoogle-glog-dev \
      libsuitesparse-dev \
      python-catkin-tools \
      ros-${ROS_DISTRO}-cv-bridge \
      ros-${ROS_DISTRO}-image-transport \
      ros-${ROS_DISTRO}-message-filters \
      ros-${ROS_DISTRO}-tf && \
      rm -rf /var/lib/apt/lists/* && \
      # Build and install Ceres
      git clone https://gitee.com/mirrors/ceres-solver && \
      cd ceres-solver && \
      git checkout tags/${CERES_VERSION} && \
      mkdir build && cd build && \
      cmake .. && \
      make -j4 install && \
      rm -rf ../../ceres-solver && \
      mkdir -p $CATKIN_WS/src/VINS-Fusion/

# Copy VINS-Fusion
COPY ./ $CATKIN_WS/src/VINS-Fusion/
# use the following line if you only have this dockerfile
# RUN git clone https://github.com/HKUST-Aerial-Robotics/VINS-Fusion.git

# Build VINS-Fusion
WORKDIR $CATKIN_WS
ENV TERM xterm
ENV PYTHONIOENCODING UTF-8
RUN catkin config \
      --extend /opt/ros/$ROS_DISTRO \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=Release && \
    catkin build && \
    sed -i '/exec "$@"/i \
            source "/root/catkin_ws/devel/setup.bash"' /ros_entrypoint.sh`

Myracle1, 2024-03-07:
> FROM osrf/ros:kinetic-desktop-full

ENV CERES_VERSION="1.12.0"
ENV CATKIN_WS=/root/catkin_ws

      # set up thread number for building
RUN   apt-get update && apt-get install -y \
      cmake \
      libatlas-base-dev \
      libeigen3-dev \
      libgoogle-glog-dev \
      libsuitesparse-dev \
      python-catkin-tools \
      ros-${ROS_DISTRO}-cv-bridge \
      ros-${ROS_DISTRO}-image-transport \
      ros-${ROS_DISTRO}-message-filters \
      ros-${ROS_DISTRO}-tf && \
      rm -rf /var/lib/apt/lists/* && \
      # Build and install Ceres
      git clone https://gitee.com/mirrors/ceres-solver && \
      cd ceres-solver && \
      git checkout tags/${CERES_VERSION} && \
      mkdir build && cd build && \
      cmake .. && \
      make -j4 install && \
      rm -rf ../../ceres-solver && \
      mkdir -p $CATKIN_WS/src/VINS-Fusion/

# Copy VINS-Fusion
COPY ./ $CATKIN_WS/src/VINS-Fusion/
# use the following line if you only have this dockerfile
# RUN git clone https://github.com/HKUST-Aerial-Robotics/VINS-Fusion.git

# Build VINS-Fusion
WORKDIR $CATKIN_WS
ENV TERM xterm
ENV PYTHONIOENCODING UTF-8
RUN catkin config \
      --extend /opt/ros/$ROS_DISTRO \
      --cmake-args \
        -DCMAKE_BUILD_TYPE=Release && \
    catkin build && \
    sed -i '/exec "$@"/i \
            source "/root/catkin_ws/devel/setup.bash"' /ros_entrypoint.sh