HKUST-Aerial-Robotics/VINS-Fusion #217: Heavy drift on combining stereo camera + IMU - ZED2

Aishkrish18, 2023-08-10:
Hey, i am encountering the exact same problem with zed2 stereo camera, when combined with IMU it has extremely heavy drift. I don't know which tuning parameters to control in order to reduce this. The timestamps and time offsets seem okay as well. Did you find a solution since you put it as completed?  

https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/assets/47241581/e4b840a6-3255-42ea-9eba-f42a6dd07fc1

knightwzh, 2024-01-25:
For me, adjust IMU's parameters could help. I slightly increase "acc_n", "gyr_n", "acc_w" and "gyr_w", and the strange drift disappears.

tulingcheng86, 2024-01-27:
> For me, adjust IMU's parameters could help. I slightly increase "acc_n", "gyr_n", "acc_w" and "gyr_w", and the strange drift disappears.

你好，请问你是怎么增加这四个值的，增加多少能改善效果，能教教我吗

tulingcheng86, 2024-01-27:
> > For me, adjust IMU's parameters could help. I slightly increase "acc_n", "gyr_n", "acc_w" and "gyr_w", and the strange drift disappears.
> 
> 你好，请问你是怎么增加这四个值的，增加多少能改善效果，能教教我吗

Hello, how do you increase these four values? How much can be increased to improve the effect?

knightwzh, 2024-03-08:
> > For me, adjust IMU's parameters could help. I slightly increase "acc_n", "gyr_n", "acc_w" and "gyr_w", and the strange drift disappears.
> 
> 你好，请问你是怎么增加这四个值的，增加多少能改善效果，能教教我吗

For my ZED2, I set acc_n=0.2, gyr_n=0.1, acc_w=0.05 and gyr_w=0.0005

tulingcheng86, 2024-03-08:
> > > 对我来说，调整 IMU 的参数可能会有所帮助。我稍微增加“acc_n”，“gyr_n”，“acc_w”和“gyr_w”，奇怪的漂移消失了。
> > 
> > 
> > 你好，请问你怎么增加这四个值的，增加多少可以改善效果，能教教我吗？
> 
> 对于我的 ZED2，我设置 acc_n=0.2、gyr_n=0.1、acc_w=0.05 和 gyr_w=0.0005

OK,thinks

tulingcheng86, 2024-03-08:
> > > 对我来说，调整 IMU 的参数可能会有所帮助。我稍微增加“acc_n”，“gyr_n”，“acc_w”和“gyr_w”，奇怪的漂移消失了。
> > 
> > 
> > 你好，请问你怎么增加这四个值的，增加多少可以改善效果，能教教我吗？
> 
> 对于我的 ZED2，我设置 acc_n=0.2、gyr_n=0.1、acc_w=0.05 和 gyr_w=0.0005

OK,THANKS