HKUST-Aerial-Robotics/VINS-Fusion #220: OutOfMemoryError on Jetson Orin Nano

Diva-Mio, 2023-09-20:
When I ran MH01_easy.bag (Vins mono / Fusion), I got the following error message:

terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 369394115439360 bytes in function 'OutOfMemoryError'
Aborted (core dumped)

How can I to solve this problem?
My Environment : 
1. Hardware:Jetson Orin Nano 8GB develop kit
2. OS:Jetpack5.1.1
3. installed opencv library:libopencv-viz4.2, python3-opencv/focal,now 4.2.0+dfsg-5 arm64

patrickpoirier51, 2023-09-29:
I confirm the same error on a Jetson Xavier NX 
 nvidia-jetpack Version: 5.0.2-b231
ROS Noetic

The error message:  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/alloc.cpp:73:
is referencing an inexsistant path on the Jetson , I suspect it  is a reference to cv-bridge within ROS

I tried different version of OpenCV and built the https://github.com/ros-perception/vision_opencv/tree/noetic vision_opencv with no succes

patrickpoirier51, 2023-11-03:
Just a quick update, Removing refs to JETPACK in sources made it installing standard Opencv 4.2  

/etc/apt/sources.list.d/nvidia-l4t-apt-source.list  
commented  (#) these lines 
deb https://repo.download.nvidia.com/jetson/common r35.4 main 
deb https://repo.download.nvidia.com/jetson/t194 r35.4 main 

sudo apt update 
sudo apt install libopencv-dev 
All 4 .2 libs installed clean 

VINS-Fusions build and works @ 15 FPS

otis-XJY, 2023-11-28:
I have the same problem in NX. Does it worked out?

patrickpoirier51, 2023-11-28:
> I have the same problem in NX. Does it worked out?

Yes

otis-XJY, 2023-11-28:
Thank you for your reply. But I'm a little confused

> Just a quick update, Removing refs to JETPACK in sources made it installing standard Opencv 4.2
>
> /etc/apt/sources.list.d/nvidia-l4t-apt-source.list commented (#) these lines deb [...]

Thank you for your reply. Please forgive my broken English.
I'm a little confused about your Opencv version. The error in NX is about opencv4.5.4. But you removed opencv 4.2 and resolve the problem.
And how did you remove opencv4.2? just "commented (#) these lines"  as you said? 
Or you did like this
sudo apt-get purge libopencv*
sudo apt autoremove
sudo apt-get update

Diva-Mio, 2023-11-29:
https://forums.developer.nvidia.com/t/compiling-opencv-on-jetpack-5/219668

This is a solve about clean opencv4.5.4 and 4.2.
I run this .sh after install ROS.
It can work for me.
And it has a few problem between build opencv4.6.0, So you need to search the solution about it.

patrickpoirier51, 2023-11-29:
Yes you have tu purge old release first = Your apt sequence is good

Then you can check if all installed packages are 4,2 with this command
apt list --installed

If you check before purge and update without jetpack, you will see a mix of 4.2 and 4.5 ; that is the root cause of the problem

otis-XJY, 2023-12-05:
> https://forums.developer.nvidia.com/t/compiling-opencv-on-jetpack-5/219668
> 
> This is a solve about clean opencv4.5.4 and 4.2. I run this .sh after install ROS. It can work for me. BTW it has a few problem between build opencv4.6.0, So you need to search the solution about it.

thank you for your advice. but I have installed the opencv4.5.3 with cuda. If I run this .sh, it will remove all the opencv package,right?

Diva-Mio, 2023-12-05:
Yes, if you don't want to install opencv 4.6.0 you can modify that file. And install opencv4.5.3 after clean.

otis-XJY, 2023-12-05:
> Yes, if you don't want to install opencv 4.6.0 you can modify that file. And install opencv4.5.3 after clean.

OK，I see. But I have other project need opencv, so it's a little risky to remove all opencv. I have tried to remove so .so by `find /usr/lib/aarch64-linux-gnu -type f -name "*4.5.4*" -exec rm {} \;`(my jetpack opencv is 4.5.4)
it seem work for me

anandvgeorge, 2024-03-21:
I have faced similar issue recently on Jetson Orin. For me the issue was resolved after I built `cv_bridge` and dependent ROS packages from source, with the same OpenCV version used for compiling VINS-Fusion.

abdul-mannan-khan, 2024-08-28:
Not sure if it helps or not but here is the solution: 

The essential software environment is same as VINS-Fusion. Besides, it requires OpenCV cuda version.(Only test it on OpenCV 4.5.3 and 4.6.0 with CUDA/CUDNN), and it have to rebuild the cv_bridge after install the OpenCV4 so that the OutOfMemoryError will not happend. For opencv, add the path of the OpenCVConfig.cmake file and cv_bridgeConfig.cmake file to your following packages VINS-Fusion/camera_models/CMakeLists.txt (change Line 16), VINS-Fusion/loop_fusion/CMakeLists.txt (Line 19) and VINS-Fusion/vins_estimator/CMakeLists.txt (Line 9 and 19) by adding following lines right before find_package(OpenCV 4 REQUIRED) as:

set(OpenCV_DIR /home/mannan/Downloads/workspace/opencv-4.6.0/release) #/home/mannan/Downloads is where your [OpenCV 4.6.0 sh](https://github.com/AastaNV/JEP/blob/master/script/install_opencv4.6.0_Jetson.sh) file was placed. You need to modify it according to your computer name
set(cv_bridge_DIR /opt/ros/noetic/share/cv_bridge/cmake)
set(cv_bridge_DIR /home/mannan/Sysal-Autonomous-Drone/devel/share/cv_bridge/cmake)