HKUST-Aerial-Robotics/VINS-Fusion #221: Mono + IMU  - Realsense d455 shows nothing on ubuntu 20.04 Orin NX

Hendyley, 2023-10-05:
I am following the instruction from

https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/
 
I successfully catkin_make

however when I try to run 

[3.1 Monocualr camera + IMU](https://github.com/HKUST-Aerial-Robotics/VINS-Fusion/#31-monocualr-camera--imu)
    roslaunch vins vins_rviz.launch
    rosrun vins vins_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_mono_imu_config.yaml 
    (optional) rosrun loop_fusion loop_fusion_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_mono_imu_config.yaml 
    rosbag play MH_01_easy.bag
    
    
 I got error 
 
 terminate called after throwing an instance of 'cv::Exception'
  what():  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/matrix.cpp:250: error: (-215:Assertion failed) s >= 0 in function 'setSize'

libobo6, 2024-02-17:
I got the same error  in ubuntu 20.04, Jetson  Xavier NX.  Is there a good solution currently available？

lijiaming5550, 2024-04-01:
同样问题

lijiaming5550, 2024-04-03:
该问题是由于cv_bridge与opencv版本冲突导致的：例如本机采用的opencv版本为4.6.0，而采用apt-get install cv_bridge的对应版本为4.2.0.
解决方案为：将cv_bridge并链接到自己的opencv版本，可参考csdn相关博客

lijiaming5550, 2024-04-03:
该问题是由于cv_bridge与opencv版本冲突导致的：例如本机采用的opencv版本为4.6.0，而采用apt-get install cv_bridge的对应版本为4.2.0.
解决方案为：将cv_bridge并链接到自己的opencv版本，可参考csdn相关博客

This issue is caused by a conflict between the cv_bridge and opencv versions: for example, the opencv version used on this machine is 4.6.0, while the corresponding version of apt-get install cv_bridge is 4.2.0.
The solution is to link cv_bridge to your own opencv version, which can be referred to on CSDN related blogs