HKUST-Aerial-Robotics/VINS-Fusion #228: D435i wrong in rviz

cwahaha51, 2024-01-10:
[image]

cwahaha51, 2024-01-10:
roslaunch vins vins_rviz.launch 

[ERROR] [1704862146.656131470]: Ignoring transform for child_frame_id "body" from authority "unknown_publisher" because of a nan value in the transform (-nan -nan -nan) (-nan -nan -nan -nan)
[ERROR] [1704862146.656158407]: Ignoring transform for child_frame_id "body" from authority "unknown_publisher" because of an invalid quaternion in the transform (-nan -nan -nan -nan)

closed 2024-01-11 by cwahaha51

srendit, 2024-03-30:
I meet the same problem. What do you do to slove the problem.

srendit, 2024-03-30:
I meet the same problem. What do you do to slove the problem ?