HKUST-Aerial-Robotics/VINS-Fusion #229: Hello

ZW-PRO, 2024-01-18:
Has anyone tested only stero, and how do you get body_T_cam0 and body_T_cam1?

tulingcheng86, 2024-01-24:
I have same problem

ZW-PRO, 2024-01-25:
> I have same problem

I tried running ROS2 in Ubuntu20.04 and did not have any problems with increasing memory. And when I run it in an embedded device, there's an issue of increasing memory. After my testing, I found that in the image_callback() function, if I impose a queue size limit, memory will not increase all the time, but this may cause image loss issues.My guess is that in embedded devices programs can process images faster than they can read them into the cache, causing memory to keep growing.

ZW-PRO, 2024-01-25:
> I have same problem

you only need to satisfy cam1_T_cam0 = body_T_cam0*(body_T_cam1)^-1

tulingcheng86, 2024-01-25:
> > I have same problem
> 
> you only need to satisfy cam1_T_cam0 = body_T_cam0*(body_T_cam1)^-1

THINKS!But iI just have the following parameters：

[front] leftIntrinsics = {240.000000, 0.000000, 240.000000, 0.000000, 240.000000, 240.000015, 0.000000, 0.000000, 1.000000 }
[front] rightIntrinsics = {240.000000, 0.000000, 240.000000, 0.000000, 240.000000, 240.000015, 0.000000, 0.000000, 1.000000 }
[front] rotationLeftInRight = {1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000 }
[front] translationLeftInRight = {-63.780090, 0.000000, 0.000000 }

so i just have ：
cam1_T_cam0 = [RotationLeftInRight | translationLeftInRight]

it  can't let me figure out   body_T_cam0 and body_T_cam1  (｡•́︿•̀｡)

ZW-PRO, 2024-01-25:
you can set body_T_cam0 = Identity matrix，so body_T_cam1 = cam0_T_cam1

tulingcheng86, 2024-01-25:
> you can set body_T_cam0 = Identity matrix，so body_T_cam1 = cam0_T_cam1

THINKS!I REALLY APPRECIATE IT.

ZW-PRO, 2024-01-25:
> > you can set body_T_cam0 = Identity matrix，so body_T_cam1 = cam0_T_cam1
> 
> THINKS!I REALLY APPRECIATE IT.

cam1_T_cam0 = [RotationLeftInRight | translationLeftInRight]，，or  cam0_T_cam1 = [RotationLeftInRight | translationLeftInRight] ？

tulingcheng86, 2024-01-25:
> > > you can set body_T_cam0 = Identity matrix，so body_T_cam1 = cam0_T_cam1
> > 
> > 
> > THINKS!I REALLY APPRECIATE IT.
> 
> cam1_T_cam0 = [RotationLeftInRight | translationLeftInRight]，，or cam0_T_cam1 = [RotationLeftInRight | translationLeftInRight] ？

cam0_T_cam1
Leftcamera is cam0.

ZW-PRO, 2024-01-25:
> > > > you can set body_T_cam0 = Identity matrix，so body_T_cam1 = cam0_T_cam1
> > > 
> > > 
> > > THINKS!I REALLY APPRECIATE IT.
> > 
> > 
> > cam1_T_cam0 = [RotationLeftInRight | translationLeftInRight]，，or cam0_T_cam1 = [RotationLeftInRight | translationLeftInRight] ？
> 
> cam0_T_cam1 Leftcamera is cam0.

OK

ZW-PRO, 2024-01-26:
May I ask about the performance of your VINS-Fusion with a pure stereo setup？

tulingcheng86, 2024-01-26:
> May I ask about the performance of your VINS-Fusion with a pure stereo setup？

I haven't calibrated the binocular camera + IMU yet. I only have the internal parameters of my camera, and I fill in the body_to_cam as described above, and I can't achieve trajectory estimation. But I can achieve binocular trajectory estimation with XTDRONE's body_to_cam parameters, but not with IMU (severe drift). So I'm now learning the binocular + IMU calibration process.

tulingcheng86, 2024-01-26:
just   stereo
[image]

ZW-PRO, 2024-01-26:
Can I add you a wechat？

tulingcheng86, 2024-01-26:
> Can I add you a wechat？

ok，17792506939