HKUST-Aerial-Robotics/VINS-Fusion #235: issue about boost

Yx-QI, 2024-03-28:
/usr/bin/ld: CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o: in function `void boost::sp_adl_block::intrusive_ptr_release<boost::filesystem::detail::dir_itr_imp, boost::sp_adl_block::thread_safe_counter>(boost::sp_adl_block::intrusive_ref_counter<boost::filesystem::detail::dir_itr_imp, boost::sp_adl_block::thread_safe_counter> const*)':
intrinsic_calib.cc:(.text._ZN5boost12sp_adl_block21intrusive_ptr_releaseINS_10filesystem6detail11dir_itr_impENS0_19thread_safe_counterEEEvPKNS0_21intrusive_ref_counterIT_T0_EE[_ZN5boost12sp_adl_block21intrusive_ptr_releaseINS_10filesystem6detail11dir_itr_impENS0_19thread_safe_counterEEEvPKNS0_21intrusive_ref_counterIT_T0_EE]+0x25): undefined reference to `boost::filesystem::detail::dir_itr_imp::~dir_itr_imp()'
/usr/bin/ld: intrinsic_calib.cc:(.text._ZN5boost12sp_adl_block21intrusive_ptr_releaseINS_10filesystem6detail11dir_itr_impENS0_19thread_safe_counterEEEvPKNS0_21intrusive_ref_counterIT_T0_EE[_ZN5boost12sp_adl_block21intrusive_ptr_releaseINS_10filesystem6detail11dir_itr_impENS0_19thread_safe_counterEEEvPKNS0_21intrusive_ref_counterIT_T0_EE]+0x2e): undefined reference to `boost::filesystem::detail::dir_itr_imp::operator delete(void*)'
/usr/bin/ld: CMakeFiles/Calibrations.dir/src/intrinsic_calib.cc.o: in function `main':
intrinsic_calib.cc:(.text.startup+0x1235): undefined reference to `boost::filesystem::detail::directory_iterator_construct(boost::filesystem::directory_iterator&, boost::filesystem::path const&, unsigned int, boost::filesystem::detail::directory_iterator_params*, boost::system::error_code*)'
/usr/bin/ld: intrinsic_calib.cc:(.text.startup+0x1276): undefined reference to `boost::filesystem::directory_entry::get_status(boost::system::error_code*) const'
/usr/bin/ld: intrinsic_calib.cc:(.text.startup+0x1f6d): undefined reference to `boost::filesystem::detail::path_algorithms::filename_v3(boost::filesystem::path const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/build.make:390：/home/qyx/1_ws/devel/lib/camera_models/Calibrations] 错误 1
make[1]: *** [CMakeFiles/Makefile2:908：VINS-Fusion/camera_models/CMakeFiles/Calibrations.dir/all] 错误 2
make[1]: *** 正在等待未完成的任务....

Assuming I already have boost installed, How can I solve this error?

wells-sde, 2024-11-14:
遇到同样的问题

JakeFishcode, 2024-11-30:
有解决了吗？

zhikai0, 2025-02-19:
我在orin 22.04 docker镜像也遇到了，尝试手动添加Boost的一些依赖，依旧报错，但是vins只依赖camera_models，该模块是可以正常编译的，希望可以帮到你。