HKUST-Aerial-Robotics/VINS-Fusion #236: some problem about opencv

zhendebufapyq, 2024-03-28:
[image]
[image]
Whether using EuRoC Example or D455 camera, this error will occur. I cannot view the opencv version using the pkg config -- modversion opencv command, but jtop can still see it (I use it on orin nx)

dahexiangdong6, 2024-07-02:
same problem!! 
you can pkg-config --modversion opencv4 view the opencv version
but I can't solve this problem

deepConnectionism, 2024-07-24:
You may need to install opencv version 3.4
@dahexiangdong6 @zhendebufapyq