HKUST-Aerial-Robotics/VINS-Fusion #239: In RVIZ the VIO trajectory is perpendicular to the gps route of global_fusion

Lmy971109, 2024-04-26:
Dear authors, thank you for your great contribution to vins-fusion! Following the (Stereo + GPS) tutorial in the readme, I downloaded the 2011_09_30_drive_0020_sync data set for testing, and used "kitti_09_30_config.yaml" as the configuration file. The commands to open the three terminals are:
(1)roslaunch vins vins_rviz.launch
(2) rosrun vins kitti_gps_test ~/catkin_vinsfusion/src/VINS-Fusion-master/config/kitti_raw/kitti_09_30_config.yaml /home/lmy/EuROC-Dataset/2011_09_30_drive_0020_sync/
(3)rosrun global_fusion global_fusion_node
However, the trajectory displayed in rviz is as shown below. It feels vertical. What is the reason? Looking forward to your reply, thank you very much!

[image]

gitshitou, 2024-11-07:
i have the same problem with you. did you have saved the problem?

hqquyen, 2025-12-19:
I have the same problem with you. did you have resolved the problem?