HKUST-Aerial-Robotics/VINS-Fusion #246: numerical unstable in preintegration

Neoyning, 2024-05-28:
Hi there,

I am testing the VINS-Fusion with stereo inertial config, the code run properly and the GUI (rivz) is showing the correct results, but it always suddenly goes into warning "numerical unstable in preintegration", and from this on the estimation of trajectory diverges.

[image]

`[ WARN] [1716886222.055879229]: numerical unstable in preintegration
[ WARN] [1716886222.057814121]: numerical unstable in preintegration
[ WARN] [1716886222.059318981]: numerical unstable in preintegration`

This phenomenon does not always happen on the same rosbag. When I use VINS-Mono for the same rosbag, this issue is never triggered and shows robust performance. So now I only use mono inertial instead of stereo inertial when testing VINS. 

Does anyone know the reason for this phenomenon? 

Thanks

Neoyning, 2024-05-28:
The "numerical unstable in preintegration" happens when computing the Jacobian of IMU preintegration. 

`if (jacobian_pose_i.maxCoeff() > 1e8 || jacobian_pose_i.minCoeff() < -1e8)
{
    ROS_WARN("numerical unstable in preintegration");
}
`

ZW-PRO, 2024-09-04:
Pls have you solved it？

wells-sde, 2024-12-02:
> The "numerical unstable in preintegration" happens when computing the Jacobian of IMU preintegration.
> 
> `if (jacobian_pose_i.maxCoeff() > 1e8 || jacobian_pose_i.minCoeff() < -1e8) { ROS_WARN("numerical unstable in preintegration"); } `

encounter the same problem, have you soleved this?