HKUST-Aerial-Robotics/VINS-Fusion #254: rosbag playback, but loop_fusion_node doesn't work as vins_node does. it seems stucked.

lida2003, 2024-09-15:
Tried below commands, but loop_fusion_node doesn't work as vins_node do, any idea?

```
    roslaunch vins vins_rviz.launch
    rosrun loop_fusion loop_fusion_node ~/catkin_ws/src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml 
    rosbag play MH/MH_01_easy.bag
```

```
$ rosrun loop_fusion loop_fusion_node src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml
config_file: src/VINS-Fusion/config/euroc/euroc_stereo_imu_config.yaml
vocabulary_file/home/daniel/ego-planner/src/VINS-Fusion/loop_fusion/../support_files/brief_k10L6.bin
loop start load vocabulary
BRIEF_PATTERN_FILE/home/daniel/ego-planner/src/VINS-Fusion/loop_fusion/../support_files/brief_pattern.yml
cam calib path: src/VINS-Fusion/config/euroc/cam0_mei.yaml
VIO input, perfrom 4 DoF (x, y, z, yaw) pose graph optimization
no previous pose graph
[ WARN] [1726401233.475202402]: image discontinue! detect a new sequence!
new sequence
sequence cnt 2
[ WARN] [1726401260.584719800]: image discontinue! detect a new sequence!
new sequence
sequence cnt 3
```

lida2003, 2024-09-16:
My mistake, it's OK now.

closed 2024-09-16 by lida2003

jysh1023, 2025-01-15:
I have the same problem. Could you share what was the issue?

lida2003, 2025-01-15:
@jysh1023 [Linux 35.5 + JetPack v5.1.3@VINS-Fusion编译安装](https://blog.csdn.net/lida2003/article/details/142204809)