HKUST-Aerial-Robotics/VINS-Fusion #256: IMU data seems NOT properly working in VINS-Fusion algorithm

lida2003, 2024-09-21:
Hi

I have got IMU data, but VINS-Fusion printf "wait for imu ...". 

Is there any configuration I should change?

My setup: Mono Camera + IMU 

```
daniel@daniel-nvidia:~/ego-planner$ rosrun vins vins_node src/VINS-Fusion/config/uav_bridge/mono_imu_config.yaml
[ INFO] [1726886998.051796615]: init begins
config_file: src/VINS-Fusion/config/uav_bridge/mono_imu_config.yaml
USE_IMU: 1
IMU_TOPIC: /tmp/uav_imu
result path ~/output//vio.csv
[ WARN] [1726886998.073582860]:  fix extrinsic param
camera number 1
[ INFO] [1726886998.075300656]: Synchronized sensors, fix time offset: 0
[ INFO] [1726886998.075343026]: ROW: 480 COL: 752
 exitrinsic cam 0
 0.0148655  -0.999881  0.0041403
  0.999557  0.0149672  0.0257155
-0.0257744 0.00375619   0.999661
-0.0216401  -0.064677 0.00981073
set g       0       0 9.81007
[ INFO] [1726886998.075494360]: reading paramerter of camera src/VINS-Fusion/config/uav_bridge/cam0_mei.yaml
MULTIPLE_THREAD is 1
[ WARN] [1726886998.076127249]: waiting for image and imu...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
wait for imu ...
terminate called without an active exception
^CAborted (core dumped)

daniel@daniel-nvidia:~/ego-planner$ rostopic hz /tmp/uav_imu
subscribed to [/tmp/uav_imu]
average rate: 99.943
        min: 0.000s max: 0.055s std dev: 0.01067s window: 99
average rate: 99.913
        min: 0.000s max: 0.055s std dev: 0.00995s window: 200
^Caverage rate: 99.801
        min: 0.000s max: 0.064s std dev: 0.01046s window: 252
daniel@daniel-nvidia:~/ego-planner$ rostopic echo /tmp/uav_imu
header:
  seq: 4319
  stamp:
    secs: 1723960858
    nsecs: 785055000
  frame_id: "world"
orientation:
  x: -0.003939514048397541
  y: 0.00873828399926424
  z: 0.17272813618183136
  w: 0.9849229454994202
orientation_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
angular_velocity:
  x: -0.003097458044067025
  y: 0.0013814261183142662
  z: 0.0009629281121306121
angular_velocity_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
linear_acceleration:
  x: 0.11278050392866135
  y: 0.0505916029214859
  z: -9.793631553649902
linear_acceleration_covariance: [0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
---
```

closed 2024-09-22 by lida2003

camron-BHF, 2024-10-01:
Did you solve this issue? If so what was the solution?

lida2003, 2024-10-01:
@camron-BHF 
With hands on the algorithm, I thought it's heavily depended on sensor data timing. 
The above issue I have met is timing issue.

camron-BHF, 2024-10-01:
@lida2003 
I'm using a realsense d455 and when I run it using intel's realsense-ros package and launch file everything works fine. But when I run the VINS package I just get 'waiting for image and imu' and nothing is being published to the topics when I subscribe to the camera topis like imu and image_rect_raw. I'm having a tough time determining the issue. I have the imu set to linear interpolation so that gyro and acceleration data are at the same timestamp.

lida2003, 2024-10-01:
1. Check ros topic with `rostopic list` // if there is any topic advertised
2. Then `rostopic echo blabla...`  //check if there is any data avaliable
3. re-route/map the topic to VINS

I'm using data directly from camera or rtp streaming and ardupilot. Maybe we have different configuration. Just check those input/ouputs. 

Hope it will help.

camron-BHF, 2024-10-01:
@lida2003 

`rostopic list` has all the necessary topics advertised but when i echo them they are blank.
Whereas when I echo them when I run the camera node using the realsense camera package they published the data correctly.
I'll try re-reouting/mapping the topic to VINS.

Thanks

camron-BHF, 2024-10-02:
@lida2003 have you ran the package through a docker container? I'm having issues with it and seems like there are some typo's in the dockerfile as its calling the docker command while already inside a container. Wanna know if I'm doing something wrong or other's have encountered this issue

lida2003, 2024-10-02:
> have you ran the package through a docker container?

No

> I'm having issues with it and seems like there are some typo's in the dockerfile as its calling the docker command while already inside a container. Wanna know if I'm doing something wrong or other's have encountered this issue

fire an issue if possible, which might be get more help from others.